STMicroelectronics/STM32_open_pin_data #2: STM32L1: GPIO IP files define GPIO_AF10_USB while it should not

fpistm, 2021-07-12:
Hi @CMOSTM 

The `STM32L152RETx.xml` defined the GPIO configFile version as: `STM32L162xE_gpio_v1_0`
https://github.com/STMicroelectronics/STM32_open_pin_data/blob/d1b80258bc63add9baad70be0ef74f3c4301ac58/mcu/STM32L152RETx.xml#L54

In `GPIO-STM32L162xE_gpio_v1_0_Modes.xml`,  `PA11` (USB DM) and `PA12` (USB DP) define an `GPIO_AF10_USB` as possible value:
https://github.com/STMicroelectronics/STM32_open_pin_data/blob/d1b80258bc63add9baad70be0ef74f3c4301ac58/mcu/IP/GPIO-STM32L162xE_gpio_v1_0_Modes.xml#L773-L777
https://github.com/STMicroelectronics/STM32_open_pin_data/blob/d1b80258bc63add9baad70be0ef74f3c4301ac58/mcu/IP/GPIO-STM32L162xE_gpio_v1_0_Modes.xml#L798-L802

Datasheet of the `STM32L152RETx` does not references them. Moreover `GPIO_AF10_USB` does not exist  in [STM32L1xx Cube Fw](https://github.com/STMicroelectronics/STM32CubeL1/search?q=GPIO_AF10_USB) only in STM32L5 and STM32WB.
https://www.st.com/resource/en/datasheet/stm32l152re.pdf

[image]

So all STM32L1 GPIO IP files should not contain it:
```Log
GPIO-STM32L152x8_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L152x8_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L152xC_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L152xC_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L15xxA_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L15xxA_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L162xD_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L162xD_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L162xE_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
GPIO-STM32L162xE_gpio_v1_0_Modes.xml:                <PossibleValue>GPIO_AF10_USB</PossibleValue>
```

CMOSTM, 2021-07-12:
Hi Frederic

Ok thanks to report this issue, I will raise an issue in our Bugzilla DB to fix it.
I will inform you

BR

CMO

CMOSTM, 2021-12-03:
https://intbugzilla.st.com/show_bug.cgi?id=110512 is not yet implemented in STM32CubeMX6.4.0 ;-(

CMOSTM, 2022-01-19:
https://intbugzilla.st.com/show_bug.cgi?id=110512 is closed as invalid for the reason explained in last comment of this ticket

fpistm, 2022-01-19:
Hi @CMOSTM 
I'm not agree with this analysis. See my comment in the ticket. There is a misalignment somewhere.
How the STM32L1 GPIO IP files could define an AF which does not exist in the STM32L1 HAL?

CMOSTM, 2022-01-20:
Ok @fpistm 
https://intbugzilla.st.com/show_bug.cgi?id=110512 is reopened 
Sorry for the misunderstanding

fpistm, 2022-07-05:
Hi @CMOSTM 

When it will be fixed ?
Only one file have been updated in the last update. Is there any issue?

fpistm, 2022-11-30:
Fixed within the tag https://github.com/STMicroelectronics/STM32_open_pin_data/releases/tag/STM32CubeMX-DB.6.0.70

Thanks @CMOSTM for your support.

closed 2022-11-30 by fpistm