STMicroelectronics/STM32_open_pin_data #7: Consider adding all possible tables from datasheets and reference manuals

Samo1289, 2022-08-29:
Hello,

Codegen and compile time static analysis is very desirable feature to have and many nice structures can be generated, to help programmers correctly tie together peripherals (TIM to peripheral, DMA to peripheral, RCC to peripheral, etc).

Thank you.

salkinium, 2022-08-29:
I've [written my thesis on this topic](https://salkinium.com/master.pdf), and it turns out to be non-trivial to generate a better SVD file from the technical docs alone.
But perhaps merging the CMSIS header files with the reference manuals may be a good option for fidelity and resolution.

Samo1289, 2022-08-29:
@salkinium  Thank you very much for the paper, I will read it during the weekend. Generating might be non-trivial, but maybe we can consider human resources; retrospectively fitting all the data into the SVD files is a tedious, boring job but it can be done; while all the new products from vendors (not only ST) could have the engineers writing the datasheet, generate the SVD files (this can also be automated quite well)

salkinium, 2022-08-29:
We want the same thing, I'm just tired of waiting for it to happen 😜
We can ask for more data, but I don't think it'll happen anytime soon, it's just difficult to sell to hardware companies that don't understand the new embedded software world.

By parsing the PDFs directly, we can circumvent any vendor gatekeeping and even retroactively apply this method to deprecated documentation as well.
The vendors are then simply not part of this discussion anymore, we don't need them to cooperate and they cannot stop us.