STMicroelectronics/STM32_open_pin_data #8: GPIO-STM32F051 AF correction

Bytegrity, 2023-01-31:
Hi there,

While performing some file generation operations on a [repo](https://github.com/zephyrproject-rtos/hal_stm32) reliant on this one. I noticed differences caused by a changed alternate function index. I managed to trace down the source causing the unexpected output to https://github.com/STMicroelectronics/STM32_open_pin_data/blob/6f7f7a0a4d6ef31b94a63b85932db17118499ea0/mcu/IP/GPIO-STM32F051_gpio_v1_0_Modes.xml#L2251

I verified a couple of related MCU datasheets and I believe the change below may be required:
```
--- a/mcu/IP/GPIO-STM32F051_gpio_v1_0_Modes.xml
+++ b/mcu/IP/GPIO-STM32F051_gpio_v1_0_Modes.xml
@@ -2248,7 +2248,7 @@
         </SpecificParameter>
         <PinSignal Name="TIM3_CH3">
             <SpecificParameter Name="GPIO_AF">
-                <PossibleValue>GPIO_AF1_TIM3</PossibleValue>
+                <PossibleValue>GPIO_AF0_TIM3</PossibleValue>
             </SpecificParameter>
         </PinSignal>
     </GPIO_Pin>
```
After this change and running the generator in the other repo, the output line up.

CMOSTM, 2023-02-01:
Hi

I reported the issue internally. We will have a look.
Thanks

CMO

Bytegrity, 2023-02-08:
Hi,

Any updates on the internal discussion?

CMOSTM, 2023-02-08:
A ticket has been raised internally to do the analysis
I will inform you about the progress of this ticket.

BR

CMO

paaspaas00, 2024-01-26:
@CMOSTM any news?

CMOSTM, 2024-01-29:
Hi 
It should be fixed in the next release I hope
CMO

CMOSTM, 2024-03-20:
Fixed with the new 2024 delivery MX 6.11

closed 2024-03-20 by CMOSTM