STMicroelectronics/STM32_open_pin_data #9: Wrong USB instance name for STM32U5(3-4)5*.xml file

fpistm, 2023-03-08:
With new tag 6.0.80, new STM32U5 xml are provided for STM32U535 and STM32U545.
For USB instance name  is set to `USB` while it should be `USB_DRD_FS`

Ex:
https://github.com/STMicroelectronics/STM32_open_pin_data/blob/03c08dfa00655e60f1fc4d3ff9f5d490c8cdff80/mcu/STM32U535VETxQ.xml#LL876C70-L876C70

CMOSTM, 2023-03-08:
Thanks Frederic I will report this issue by a new Bugzilla ticket in MX

CMOSTM, 2023-07-10:
Normally it is fixed with the new delivery 6.0.90 :
$ grep -ri __HAL_RCC_USB_FS_CLK_ENABLE STM32*
STM32U535CBTx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CBTxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CBUx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CBUxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CCTx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CCTxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CCUx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CCUxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CETx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CETxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CEUx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535CEUxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535JEYxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535NCYxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535NEYxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RBIx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RBIxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RBTx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RBTxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RCIx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RCIxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RCTx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RCTxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535REIx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535REIxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RETx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535RETxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VCIx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VCIxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VCTx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VCTxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VEIx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VEIxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VETx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U535VETxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U545CETx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U545CETxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
[... cut ...]
STM32U545VEIx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U545VEIxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U545VETx.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">
STM32U545VETxQ.xml:    <IP ClockEnableMode="__HAL_RCC_USB_FS_CLK_ENABLE" InstanceName="USB_DRD_FS" Name="USB_DRD_FS" Version="usb2_v1_0_Cube">

closed 2023-07-10 by CMOSTM